cvmanjoo/RTC #5: Wrong readings from PCF8563, not at all....

woutnme, 2020-11-28:
Hi dear, my readings of ESP32 in the PCF8563 its like some overflow wrong, this are the readings every second, if check when the second its over 20, add 20 to the month, and when its over 40 add 40 to seconds and month, this repeat all time. Thanks in advance

28-11-2020 22:27:8 
28-11-2020 22:27:9 
28-11-2020 22:27:10 
28-11-2020 22:27:11 
28-11-2020 22:27:12 
28-11-2020 22:27:13 
28-11-2020 22:27:14 
28-11-2020 22:27:15 
28-11-2020 22:27:16 
28-11-2020 22:27:17 
28-11-2020 22:27:18 
28-11-2020 22:27:19 
28-31-2020 22:27:20 
28-31-2020 22:27:21 
28-31-2020 22:27:22 
28-31-2020 22:27:23 
28-31-2020 22:27:24 
28-31-2020 22:27:25 
28-31-2020 22:27:26 
28-31-2020 22:27:27 
28-31-2020 22:27:29 
28-31-2020 22:27:30 
28-31-2020 22:27:31 
28-31-2020 22:27:32 
28-31-2020 22:27:33 
28-31-2020 22:27:34 
28-31-2020 22:27:35 
28-31-2020 22:27:36 
28-31-2020 22:27:37 
28-31-2020 22:27:38 
28-31-2020 22:27:39 
68-51-2020 62:27:40 
68-51-2020 62:27:41 
68-51-2020 62:27:42 
68-51-2020 62:27:43

woutnme, 2020-11-28:
Changing in the cpp the way to change the format looks like works fine....
Example (need repeat in all gets, looking the datasheet what bits are used):

uint8_t PCF8563::getDay()
{
    uint8_t day;
    Wire.beginTransmission(PCF8563_ADDR);
    Wire.write(0x05);  // Day Register
    Wire.endTransmission();
    Wire.requestFrom(PCF8563_ADDR, 1);
    day = Wire.read();
    day = (((day>>4)&1)*10)+(day&15); /////////// This isnt in the original file
    //return (bcd2bin(day)); /////////   Comment this
    return (day);/////////// This isnt in the original file
}

cvmanjoo, 2020-11-29:
Hi woutnme,

Thanks for notifying the issue. I will look into it.

Thanks
Manjunath

cvmanjoo, 2021-06-03:
Its works just fine for me on Uno, 

 28-11-2020 22:27:26 
 28-11-2020 22:27:27 
 28-11-2020 22:27:28 
 28-11-2020 22:27:29 
 28-11-2020 22:27:30 
 28-11-2020 22:27:31 
 28-11-2020 22:27:32 
 28-11-2020 22:27:33 
 28-11-2020 22:27:34 
 28-11-2020 22:27:35 
 28-11-2020 22:27:36 
 28-11-2020 22:27:37 
 28-11-2020 22:27:38 
 28-11-2020 22:27:39 
 28-11-2020 22:27:40 
 28-11-2020 22:27:41 
 28-11-2020 22:27:42 
 28-11-2020 22:27:43 
 28-11-2020 22:27:44 
 28-11-2020 22:27:45 
 28-11-2020 22:27:46 
 28-11-2020 22:27:47 
 28-11-2020 22:27:48 
 28-11-2020 22:27:49 
 28-11-2020 22:27:50 
 28-11-2020 22:27:51 
 28-11-2020 22:27:52 
 28-11-2020 22:27:53 
 28-11-2020 22:27:54 
 28-11-2020 22:27:55 
 28-11-2020 22:27:56 
 28-11-2020 22:27:57 
 28-11-2020 22:27:58 
 28-11-2020 22:27:59 
 28-11-2020 22:28:0 
 28-11-2020 22:28:1 

Try to set time using this library and try to retrieve the time.

closed 2021-06-05 by cvmanjoo