cvmanjoo/RTC #7: Arduino Nano Every compiling issue

charitygrace, 2020-12-08:
Hi,

My knowledge of C++ is limited and I'm giving up trying to figure out what is wrong.  Hopefully you'll know if the library can be modified or if I can instantiate the class differently in the code or it's something else.

Here's the issue:
I am unable to instantiate the class via "static DS3231 RTC" in the Arduino IDE when I'm setup to run a DS3231 clock via a Nano Every.

Here's my super simplified code:

```
#include <RTC.h>
#include <Wire.h>

static DS3231 RTC;

void setup() {
  Serial.begin(9600);

/*
  RTC.begin();
  RTC.setTime(4, 18, 0);      // (hour, minute, second)
  RTC.setDate(22, 10, 2020);  // (day, month, year)
  */
}
 
void loop() {
/*
  Serial.print(RTC.getDay());
  Serial.print("/");
  Serial.print(RTC.getMonth());
  Serial.print("/");
  Serial.print(RTC.getYear());
  Serial.print(" - ");
  Serial.print(RTC.getWeek());
  Serial.println();
  delay(1000);
  */
}
```

The error in my rtcV3.ino file is:

```
rtcV3:4:15: error: expected ')' before '*' token
 static DS3231 RTC;
               ^
rtcV3:4:15: error: expected ')' before '*' token
Using library RTC at version 1.0.1 in folder: /Users/charity/Documents/Arduino/libraries/RTC 
Using library Wire at version 1.0 in folder: /Users/charity/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/libraries/Wire 
exit status 1
expected ')' before '*' token
```

The Arduino IDE is setup with:
Board > "Arduino MegaAVR Boards" > "Arduino Nano Every"
Registers Emulation > "None (ATMEGA4809)"

The issue happens during compiling, it doesn't even get to upload (you can see the issue without having a board connected). Do you know what needs to be done to fix this?  Is this an Arduino IDE issue or something that needs to be tweaked in the library?

Thanks. I like your library.

cvmanjoo, 2020-12-11:
Looks likes this is not issue of this library. 

Let me know if you could fix this.

charitygrace, 2020-12-11:
Would love a little more help than that if possible since I'm not a C++ person.

Being that it's the syntax of your instantiate code that's causing the compile to fail (other libraries are working for me), I have a few ideas of where the issue is coming from:

1. the compiler is expecting something more in your DS3231 class code (either in the Public or Private part), something "more standard C++", say code with a () and a * to insatiate the class using the static method (I've tried different stackexchange examples, but nada so far)
2. the Arduino IDE for the Nano Every is just compiling things wrong and I should talk to that group about how the code is dealing with the static method
3. the Nano Every simply won't work with your library

Any thoughts on which path it might be?

cvmanjoo, 2020-12-13:
Ok,

I dont have Nano Every now. Still i will check with code and compiling,

You try to get help from Arduino forums.

I strongly believe it should work.
Lets look into it.

cvmanjoo, 2025-09-10:
Closing this by assuming issue is resolved as there is no reply.

closed 2025-09-10 by cvmanjoo