cvmanjoo/RTC #12: Does this work on ESP8266 ?

kiranshashiny, 2022-09-11:
I am trying to compile it on Wemos D1 R1 .

WARNING: library RTC-1.6.0 claims to run on avr, mbed_nano, mbed_rp2040 architecture(s) and may be incompatible with your current board which runs on esp8266 architecture(s).
In file included from /Users/shashikiran/Downloads/RTC-1.6.0/examples/Auto_setTime/Auto_setTime.ino:4:
/Users/shashikiran/Documents/Arduino/libraries/RTC-1.6.0/src/I2C_RTC.h:51:7: error: redefinition of 'class DateTime'
   51 | class DateTime {
      |       ^~~~~~~~

cvmanjoo, 2022-09-14:
It should work, I haven't tested on ESPs, so didn't add to the list.
Let me know if it works.

funkdr, 2023-03-19:
I was able to get it to work with ESP8266 NodeMCU after addressing the errors in DS3231.cpp:
changing the 'else if (h_mode == CLOCK_H12)' in the getHours() routine to just 'else', 
adding the definition '#define UNIX_OFFSET 946684800'.

There are similar errors in the other DSxxxx.cpp files, but since i am only using the DS3231 i just deleted those files from the src folder.

cvmanjoo, 2025-09-10:
This issue has been resolved.