cvmanjoo/RTC #16: epoch problem with RTC DS3231 & Nano

VenatorOblivium, 2023-05-01:
Hi. I have obtained recently the RTC DS3231, and while testing it, I get a strange error.
I don't know if is something that I'm doing wrong or some kind of incompatibility with the device.
The code that gives me problem is the following:
```c++
time_t epoch = obtainEpochFromNTP();
Serial.print(F("Unix time = "));
Serial.println(epoch);
RTC.setEpoch(epoch);
RTC.startClock();
epoch = RTC.getEpoch();
Serial.print(F("Clock Unix time = "));
Serial.println(epoch);
``` 

Before doing this, I call `RTC.begin();`, and get a true as reponse.
Now, the error is with the resulting logs:

> Unix time = 1682982775
Clock Unix time = 2629736309
Current time: 1923/5/1 23:12:55

The last line is what I get after obtaining the "current time" (from the device).

So, the first error I see is: Why, if I pass one value to setEpoch (in this case 1682982775), why I get other value when I call getEpoch (2629736309).
The second one (I'm sure is related to the first one in some degree) is that 2629736309 should be the year 2053, but I get 1923.
I'm using Arduino Nano (Chinese clon), and I have tested with 3 different RTC DS3231 modules, and obtain the same results.

tuomas2, 2023-07-23:
Similar issue on Arduino Mega board.

emwtur, 2025-09-08:
have similar issue with raspberry pico and DS3231 
Setting time using setEpoch() sets it one day wrong ?
And I have to set as_unix_epoch to true when setting time and false when reading to get it to work 
 RTC.setEpoch(epoch - SECS_PER_DAY, true)
 epoch = RTC.getEpoch(false);