cvmanjoo/RTC #18: getEpoch() returns wrong epoch on Arduino NANO

nicknugget, 2023-08-31:
When calling getEpoch it returns an Epoch that is 36 years (!) in the future, despite date and time are set and shown correctly.

Simple test code:

```
#include <Arduino.h>
#include <I2C_RTC.h>

static PCF8563 RTC;

void setup(void) {
  Serial.begin(9600);
  while (!Serial)
    ;

  // Start RTC and check status
  RTC.begin();
  if (!RTC.isRunning()) {
    RTC.setDateTime(__DATE__, __TIME__);
    RTC.updateWeek();    
  }
}

void loop(void) {
  if (RTC.isRunning()) {
    Serial.print("Time: ");
    Serial.print(RTC.getHours());
    Serial.print(":");
    Serial.print(RTC.getMinutes());
    Serial.println();
    Serial.print("Date: ");
    Serial.print(RTC.getDay());
    Serial.print("-");
    Serial.print(RTC.getMonth());
    Serial.print("-");
    Serial.print(RTC.getYear());
    Serial.println();
    Serial.print("Epoch: ");
    Serial.print(RTC.getEpoch());
    Serial.println();
    delay(1000);
  }
}
```

Output:

```
Time: 9:35
Date: 31-8-2023
Epoch: 2832768244
```

cvmanjoo, 2024-03-05:
Which nano board exactly?

DavidMS51, 2024-05-14:
I am finding this problem with a 'generic' v3 nano clone

cvmanjoo, 2024-05-16:
Arduino nano uses Atmega328p itself, right?

DavidMS51, 2024-05-16:
yes that is correct

cvmanjoo, 2024-12-06:
Fixed in v1.9.0