cvmanjoo/RTC #22: Not compatible with esp32s and NodeMcu-32s

soontai, 2024-01-30:
url = https://github.com/cvmanjoo/RTC
version=1.0.6
Not compatible with esp32s and NodeMcu-32s, hope to support esp32s and NodeMcu-32s

version = 1.0.2
A long time ago, I could compile successfully using RTC 1.0.2.
After I installed RTC 1.0.2 on other computers, I could not accept the syntax of static PCF8563 RTC.
I tried several computers and got the same error.

May I ask what is the reason?

Win10 + arduino ide 2.2.1 + rtc example pcf8563_info.ino
Board: ESP32S2 (ide board is NODEMCU-32S)

cvmanjoo, 2024-03-04:
Could you please post, what is the error?

soontai, 2024-03-05:
> Could you please post, what is the error?

#================================================
for 1.0.6 , error message :  Why I select Example PCF8563, But  use DS1307 ?
WARNING: library RTC claims to run on avr, mbed_nano, mbed_rp2040 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
c:\Users\user_dir\Documents\Arduino\libraries\RTC\src\DS1307.cpp: In member function 'void DS1307::setEpoch(time_t, bool)':
c:\Users\user_dir\Documents\Arduino\libraries\RTC\src\DS1307.cpp:504:19: error: 'UNIX_OFFSET' was not declared in this scope
   epoch = epoch - UNIX_OFFSET;
                   ^~~~~~~~~~~
c:\Users\user_dir\Documents\Arduino\libraries\RTC\src\DS1307.cpp: In member function 'time_t DS1307::getEpoch(bool)':
c:\Users\user_dir\Documents\Arduino\libraries\RTC\src\DS1307.cpp:535:12: error: 'UNIX_OFFSET' was not declared in this scope
   epoch += UNIX_OFFSET;
            ^~~~~~~~~~~
c:\Users\user_dir\Documents\Arduino\libraries\RTC\src\DS3231.cpp: In member function 'void DS3231::setEpoch(time_t, bool)':
c:\Users\user_dir\Documents\Arduino\libraries\RTC\src\DS3231.cpp:614:19: error: 'UNIX_OFFSET' was not declared in this scope
   epoch = epoch - UNIX_OFFSET;
                   ^~~~~~~~~~~
c:\Users\user_dir\Documents\Arduino\libraries\RTC\src\DS3231.cpp: In member function 'time_t DS3231::getEpoch(bool)':
c:\Users\user_dir\Documents\Arduino\libraries\RTC\src\DS3231.cpp:750:12: error: 'UNIX_OFFSET' was not declared in this scope
   epoch += UNIX_OFFSET;
            ^~~~~~~~~~~

exit status 1

Compilation error: exit status 1

#================================================
for  1.0.2 error message :  
C:\Users\user_dir\AppData\Local\Temp\.arduinoIDE-unsaved202425-3312-y1kwmu.98gvl\PCF8563_Info\PCF8563_Info.ino:4:8: error: 'PCF8563' does not name a type
 static PCF8563 RTC;
        ^~~~~~~
C:\Users\user_dir\AppData\Local\Temp\.arduinoIDE-unsaved202425-3312-y1kwmu.98gvl\PCF8563_Info\PCF8563_Info.ino: In function 'void setup()':
C:\Users\user_dir\AppData\Local\Temp\.arduinoIDE-unsaved202425-3312-y1kwmu.98gvl\PCF8563_Info\PCF8563_Info.ino:9:3: error: 'RTC' was not declared in this scope
   RTC.begin();
   ^~~
C:\Users\user_dir\AppData\Local\Temp\.arduinoIDE-unsaved202425-3312-y1kwmu.98gvl\PCF8563_Info\PCF8563_Info.ino: In function 'void loop()':
C:\Users\user_dir\AppData\Local\Temp\.arduinoIDE-unsaved202425-3312-y1kwmu.98gvl\PCF8563_Info\PCF8563_Info.ino:40:16: error: 'RTC' was not declared in this scope
   Serial.print(RTC.getDay());
                ^~~

exit status 1

Compilation error: 'PCF8563' does not name a type

cvmanjoo, 2024-03-05:
I got NodeMCU board and testing.
Will confirm

cvmanjoo, 2024-03-08:
Fixed this in v1.7.1

closed 2024-03-08 by cvmanjoo

soontai, 2024-03-08:
> Fixed this in v1.7.1

Test OK, Thank you very much.